duxingkei33/orangepi_PC_gpio_pyH3 #3: POWER_LED not blink

nopnop2002, 2016-03-24:
My code is below.
STATUS_LED flashes on and off.
but POWER_LED don't flashes on and off.

 -*- coding: utf-8 -*-
from pyA20.gpio import gpio
from pyA20.gpio import port
from pyA20.gpio import connector
import time

COUNT = 3
PIN1 = port.POWER_LED
PIN2 = port.STATUS_LED
#PIN1 = connector.LEDp1    # This is the same as port.POWER_LED
#PIN2 = connector.LEDp2    # This is the same as port.STATUS_LED

gpio.init()
gpio.setcfg(PIN1, gpio.OUTPUT)
gpio.setcfg(PIN2, gpio.OUTPUT)

for _ in xrange(COUNT):
    gpio.output(PIN1,True)
    gpio.output(PIN2,True)
    time.sleep(1.0)
    gpio.output(PIN1,False)
    gpio.output(PIN2,False)
    time.sleep(1.0)

Sorry my Poor English.

duxingkei33, 2016-03-25:
try this example in
orangepi_PC_gpio_pyH3\examples\blink_POWER_STATUS_PL10.py

nopnop2002, 2016-03-26:
Thank you very much for your answer.

I tried orangepi_PC_gpio_pyH3\examples\blink_POWER_STATUS_PL10.py with my OrangePi-PC.
It can't blink POWER_LED(Yellow LED).
So i tried these command.
It's work fine.
I can blink POWER_LED(Yellow LED).

standby_led is linked with PL10.
I think your cord is right,but it can't blink POWER_LED(Yellow LED).

$ sudo modprobe gpio_sunxi
$ ls -l /sys/class/gpio_sw
total 0
lrwxrwxrwx 1 root root 0 Mar 26 02:53 normal_led -> ../../devices/platform/gpio_sw.1/gpio_sw/PA15
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA0 -> ../../devices/platform/gpio_sw.9/gpio_sw/PA0
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA1 -> ../../devices/platform/gpio_sw.7/gpio_sw/PA1
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA10 -> ../../devices/platform/gpio_sw.25/gpio_sw/PA10
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA11 -> ../../devices/platform/gpio_sw.3/gpio_sw/PA11
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA12 -> ../../devices/platform/gpio_sw.2/gpio_sw/PA12
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA13 -> ../../devices/platform/gpio_sw.5/gpio_sw/PA13
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA14 -> ../../devices/platform/gpio_sw.6/gpio_sw/PA14
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA15 -> ../../devices/platform/gpio_sw.1/gpio_sw/PA15
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA18 -> ../../devices/platform/gpio_sw.20/gpio_sw/PA18
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA19 -> ../../devices/platform/gpio_sw.19/gpio_sw/PA19
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA2 -> ../../devices/platform/gpio_sw.15/gpio_sw/PA2
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA20 -> ../../devices/platform/gpio_sw.27/gpio_sw/PA20
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA21 -> ../../devices/platform/gpio_sw.18/gpio_sw/PA21
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA3 -> ../../devices/platform/gpio_sw.10/gpio_sw/PA3
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA6 -> ../../devices/platform/gpio_sw.4/gpio_sw/PA6
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA7 -> ../../devices/platform/gpio_sw.21/gpio_sw/PA7
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA8 -> ../../devices/platform/gpio_sw.22/gpio_sw/PA8
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PA9 -> ../../devices/platform/gpio_sw.24/gpio_sw/PA9
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PC0 -> ../../devices/platform/gpio_sw.13/gpio_sw/PC0
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PC1 -> ../../devices/platform/gpio_sw.14/gpio_sw/PC1
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PC2 -> ../../devices/platform/gpio_sw.16/gpio_sw/PC2
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PC3 -> ../../devices/platform/gpio_sw.17/gpio_sw/PC3
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PC4 -> ../../devices/platform/gpio_sw.11/gpio_sw/PC4
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PC7 -> ../../devices/platform/gpio_sw.12/gpio_sw/PC7
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PD14 -> ../../devices/platform/gpio_sw.8/gpio_sw/PD14
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PG6 -> ../../devices/platform/gpio_sw.28/gpio_sw/PG6
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PG7 -> ../../devices/platform/gpio_sw.29/gpio_sw/PG7
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PG8 -> ../../devices/platform/gpio_sw.23/gpio_sw/PG8
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PG9 -> ../../devices/platform/gpio_sw.26/gpio_sw/PG9
lrwxrwxrwx 1 root root 0 Mar 26 02:53 PL10 -> ../../devices/platform/gpio_sw.0/gpio_sw/PL10
lrwxrwxrwx 1 root root 0 Mar 26 02:53 standby_led -> ../../devices/platform/gpio_sw.0/gpio_sw/PL10

$ sudo sh -c "echo 1 > /sys/class/gpio_sw/normal_led/data"
$ sudo sh -c "echo 0 > /sys/class/gpio_sw/normal_led/data"
$ sudo sh -c "echo 1 > /sys/class/gpio_sw/standby_led/data"
$ sudo sh -c "echo 0 > /sys/class/gpio_sw/standby_led/data"

**Thank you very much for your useful library.**

nopnop2002, 2016-03-27:
I knew the cause.
Registers begin at 0x01C20800.

Port L have a different address. 
Registers for port L begin at 0x01F02C00.

Please see this page.

http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=674&extra=page%3D3

Thank you.