duxingkei33/orangepi_PC_gpio_pyH3 #7: port.PA15 & port.PA16

nopnop2002, 2017-03-15:
Orange Pi ZERO have port.PA15 & port.PA16.
But not define in this library,

dneyirp, 2017-03-28:
Hi

It is easy to add these to the file `/pyA20/gpio/mapping.h`.
Simply go to around line number 100 and add the entry you need.
The other lines give you the format which is basically:
`{ <quoted text port name, e.g. PA15>, SUNXI_GPA(15), <number of the header pin or just 0 if not on header> `
\
Here is the PA16 line:
`{ "PA16", SUNXI_GPA(16), 0 },`

I hope this helps.

Best regards
P

lamerjack, 2021-01-06:
Perfect! i was able to add PA15 and PA16 but how to add PL0 and PL1:
i did this on /pyA20/gpio/mapping.h
/*          {   "PA1",  SUNXI_GPA(1),   11 },
            {   "PA0",  SUNXI_GPA(0),   13   },*/
            {   "PL0",  SUNXI_GPL(0),   11 },
            {   "PL1",  SUNXI_GPL(1),   13   },
but no success :( i also try 352 instea of SUNXI_GPL(0) but still fail :(

nopnop2002, 2021-01-06:
@lamerjack 

Unlike other ports, PL0 and PL1 are not consecutive addresses.

So when using mmap, it's not easy to use.

lamerjack, 2021-01-06:
unfortunately i need to use pya20 is gpio controller...
anyway i found this post from you:
https://github.com/Jeremie-C/OrangePi.GPIO/issues/3
so tomorrow i will try it!